embedded-graphics/embedded-graphics #702: The sign of angles is incorrect

rfuest, 2022-11-06:
At the moment positive angles are interpreted as counter clockwise rotations. But this is would only be correct for a coordinate system where the y axis is pointing up. In our coordinate system, where the y axis is pointing down, positive angles should cause a clockwise rotation.

rfuest, 2022-11-28:
@jamwaffles Do you think we should change this or do you prefer to just document the way angles work in e-g?

The current implementation doesn't match the angles used by the `display-driver-hal` and `ssd1306` crates. IMO it would be beneficial if angles were consistent across the wider e-g ecosystem.

jamwaffles, 2022-11-28:
I think we should change it. Maybe I'm in the minority but to me it feels more intuitive for positive to mean clockwise.

rfuest, 2022-11-28:
> I think we should change it.

OK, I'll create a PR. The change will only affect users of `Arc` and `Sector`, which is only a smaller subset of all e-g users. But the longer we wait with this change the more users would be affected, especially if we add polar coordinates (#617) or introduce other angle types for 90° degree increment rotations (#661).

> Maybe I'm in the minority but to me it feels more intuitive for positive to mean clockwise.

I'm not sure if you are in the minority, at least for computer graphics the clockwise rotation is more common and also makes more sense mathematically.

closed 2022-12-14 by rfuest