embedded-graphics/embedded-graphics #703: Add one dimensional variants of Rectangle::resized

rfuest, 2022-11-07:
Using `Rectangle::resized` to only change either the width or height of a rectangle is a fairly common operation in my code. But the resulting code is very noisy and this could be improved by adding two new methods:
```rust
// Current code:
let bounding_box = target.bounding_box(); // Note that a temporary variable is necessary
let right = bounding_box.resized(Size::new(1, bounding_box.size.height), AnchorPoint::TopRight);

// With the new method:
let right = target.bounding_box().resized_x(1, AnchorPoint::TopRight);
```

Remaining questions:
- How should the methods be named?
  `resized_x`/`resized_y`, `resized_horizontal`/`resized_vertical`, `change_width`/`change_height`, ...
- Should we reuse `AnchorPoint` or create two new enums?
  `resized_x(width, AnchorPoint::TopLeft)`, `resized_x(width, AnchorPoint::CenterLeft)`, and `resized_x(width, AnchorPoint::BottomLeft)` would, for example, produce the same output.

  The new types could look like this and provide some methods to convert from/to `AnchorPoint`:

  ```rust
  pub enum AnchorPointX { Left, Center, Right }
  pub enum AnchorPointY { Top, Center, Bottom }
  
  impl AnchorPoint {
      pub fn from_xy(x: AnchorPointX, y: AnchorPointY) -> Self { ... }
      pub fn x(self) -> AnchorPointX { ... }
      pub fn y(self) -> AnchorPointY { ... }
  }
  ```

jamwaffles, 2022-11-07:
> How should the methods be named?

How about the 4th, secret option: `resized_width`/`resized_height`? I feel like that fits with `Size` which also uses the "width" and "height" wording.

> Should we reuse AnchorPoint or create two new enums?

Separate X/Y types would encode the intent in the type system which is nice. Reusing `AnchorPoint` for individual axes would make "half" of the given variant meaningless, e.g. resizing on X with `AnchorPoint::TopRight`, the `Top` component is ignored.

As we're going for a bit of API sugar here too, we could call the two new enums `AnchorX` and `AnchorY` to reduce chars typed slightly.

rfuest, 2022-11-07:
> How about the 4th, secret option: `resized_width`/`resized_height`? I feel like that fits with `Size` which also uses the "width" and "height" wording.

SGTM.

> Separate X/Y types would encode the intent in the type system which is nice. Reusing `AnchorPoint` for individual axes would make "half" of the given variant meaningless, e.g. resizing on X with [...]

Also SGTM. `AnchorX` and `AnchorY` is a good idea, not only because it's shorter but perhaps also more accurate, because the entire edge could be considered to be the anchor and not only a single point.