embedded-graphics/embedded-graphics #714: Linux framebuffer driver?

icefoxen, 2022-11-28:
## Description of the problem/feature request/other

Is there a driver for drawing directly on the Linux kernel framebuffer device `/dev/fb0`?  If not, do you want one, perhaps as a shim for the `framebuffer` crate or such?

Inspired by https://github.com/ggez/ggez/issues/1141 .  This issue got me wondering because I expected this to exist but couldn't find one.

jamwaffles, 2022-11-28:
Not to my knowledge, no. I took a quick look through [the reverse dependencies](https://crates.io/crates/embedded-graphics/reverse_dependencies) and didn't see mention of Linux FB stuff there. I don't know where else to look so maybe it's not implemented, but it's also worth asking in [the chatroom](https://matrix.to/#/#rust-embedded-graphics:matrix.org) (if you're partial to Matrix) to see if anyone else has already implemented it.

icefoxen, 2022-11-29:
Yeah I managed to find <https://crates.io/crates/embedded_graphics_framebuffer/0.1.1> but it's only seven commits in.  I think that's the best there is right now.  Seems an odd omission; I might try to make one of my own.  Thanks!

shulltronics, 2023-09-04:
Hi guys,

I know this conversation is from about a year ago, but I did implement this in a very rudimentary way in [my repo here](https://github.com/shulltronics/displayio-fbdev). The `DrawTarget` trait is implemented at the bottom of the library code. It's also for version 0.7 of this library.

If there is still interest I could update the code! Let me know :)

codepainters, 2024-09-26:
Just found this issue looking for the framebuffer support. 

I have a real use case for it: a Linux device that draws an image, which is then displayed by HUB75 LED matrix display. 

I know there is a driver for HUB75, but this particular hardware uses a typical China-made DVI-to-HUB75 converter and I have to keep it (the project involves replacing current software, touching the hardware is not an option).

rfuest, 2024-09-28:
A driver crate to support the Linux framebuffer would be nice to have, but I don't think anything is published on crates.io yet. We don't bundle any display drivers with embedded-graphics and rely on community contributions to support a wide range of displays.

Did you try the code by @shulltronics? Perhaps this could be the base of a more mature implementation that could be published on crates.io and added to e-g's supported driver list.

codepainters, 2024-09-28:
I only looked at it briefly - it's surprisingly simple, I will give it a try for sure.