embedded-graphics/embedded-graphics #716: Is chaining still supported?

v-morlock, 2022-12-27:
- Version of embedded-graphics in use (if applicable): 0.7.1
- Target: esp32
- If using real hardware: custom

## Description of the problem/feature request/other

I've been trying to write a display driver that's able to draw pixels without writing them to a display buffer in order to use less heap on esp32. To do this, I'd need to chain iterators into one big iterator which I'd use to write the resulting frame directly to the (e-ink) display. However, i've found that support for chaining seems to be dropped from e.g. Text - the migration guide still mentions calling .pixels, however, it seems Text doesn't implement it. Are there other ways to accomplish what I'm trying or is it planned to re-add that?

Thanks!

rfuest, 2022-12-27:
Using chaining isn't recommended, because it makes it impossible to use accelerated drawing calls, like `DrawTarget::fill_solid` or `DrawTarget::fill_contiguous`. This can significantly slow down drawing of images and rectangles.

I don't know the exact requirements for your application, but generally I would suggest to implement a `Drawable` (see the example in https://docs.rs/embedded-graphics/latest/embedded_graphics/trait.Drawable.html). The same could also be achieved without implementing the trait by just move all drawing operations into a function, which takes a `DrawTarget` as a parameter:
```rust
fn draw_frame<D: DrawTarget<Color = Rgb565>>(target: &mut D) -> Result<(), D::Error> {
    target.clear(Rgb565::BLACK)?;
    Text::new("text1", ...).draw(target)?;
    Text::new("text2", ...).draw(target)?;

    Ok(())
}
```

> However, i've found that support for chaining seems to be dropped from e.g. Text - the migration guide still mentions calling .pixels, however, it seems Text doesn't implement it. Are there other ways to accomplish what I'm trying or is it planned to re-add that?

That must an error in the migration guide. There is a related open issue about missing image pixel iterators: https://github.com/embedded-graphics/embedded-graphics/issues/609. But even if it is likely that pixel iterators for texts will be added back in the future, I don't think this is the best solution for your use case.

v-morlock, 2022-12-29:
Thanks, that sounds plausible! In fact I have already implemented a drawable, the problem was that the draw_iter method is called on every draw operation. As my display only supports fully refreshing the entire image I had to buffer the pixels in an appropriate framebuffer until all drawing operations are done. This framebuffer was initially too big for the memory of my ESP32, however I have found a solution for this in the meantime. 

My original hope was that I could combine all drawing operations into one iterator by chaining, so that I only have to call draw_iter only once per frame which would allow me to send a complete image directly to the display.

closed 2022-12-29 by v-morlock

rfuest, 2022-12-29:
I've only just recently started to look into EPDs, but I think they have a framebuffer in the controller IC and use the content of that buffer to refresh the display when an update is triggered. There should be no need for a framebuffer in the ESP32, but this could be a limitation of the display driver implementation.

Sending a single rectangular image to the display will always be faster than setting individual pixels in the device's framebuffer, but for the slow update rate of an EPD this might no be a large issue.