embedded-graphics/embedded-graphics #722: 0.8 release

jamwaffles, 2023-05-09:
Development progress has pretty much halted on e-g (for now I hope!) so even though there are still other breaking changes to come, I think it's time for a 0.8 release.

@rfuest what's the minimum amount of effort we need to put into e-g to get it ready for a release. I'm not too concerned with in-flight changes or existing issues - those can be fixed later.

Please comment or edit the following list if you know of anything specific.

TODO

- [ ] Migration guide from 0.7 to 0.8

jamwaffles, 2023-05-09:
P.s. I'm personally disregarding the [0.8 milestone](https://github.com/embedded-graphics/embedded-graphics/issues?q=is%3Aopen+is%3Aissue+milestone%3A0.8)

rfuest, 2023-05-10:
> @rfuest what's the minimum amount of effort we need to put into e-g to get it ready for a release. I'm not too concerned with in-flight changes or existing issues - those can be fixed later.

At this point I'm not sure I want to continue to contribute to e-g anymore and don't really care about the details of a possible 0.8 release.

jamwaffles, 2023-05-14:
Understood. I think we're in the same boat... Regardless, thank you very much for your contributions to e-g over the years. It's truly appreciated.

jamwaffles, 2023-05-14:
I just released 0.8.0. The list of changes didn't really warrant a migration guide - the CHANGELOG.mds are enough IMO.

closed 2023-05-14 by jamwaffles