embedded-graphics/embedded-graphics #726: Add support for 12 bit RGB444 color

cosmikwolf, 2023-07-21:
Currently, there is no support for implementing 12 bit color (RGB444 = BRG444)

The benefit of using 12 bit color would be a 25% increase in performance writing pixel data to LCDs per frame. 

Many users do not care about color depth, but would rather have a very fast refresh on a display.

pferreir, 2025-05-18:
Hi! It seems this has been merged already?

rfuest, 2025-05-18:
The `Rgb444` type itself has been added in #732, but the there are still some pieces missing to make it fully supported. At the moment `Rgb444` colors in `ImageRaw`s and `Framebuffer`s are handled as `u16`s with the upper 4 bits unused. To take advantage of the memory savings `Rgb444` provides a `RawU12` type needs to be added that is packing two 12 bit color values in 3 bytes.

catherinejones, 2026-06-14:
Hello! I'm the original author of #732, I've just opened #815 to resolve this issue, sorry about the extremely long wait!