embedded-graphics/embedded-graphics #742: The style function

Millan29, 2024-01-17:
- Version of embedded-graphics in use (if applicable): [0.8.1]
- Target: [wokwi esp32c3] / [real hardware]
- If using real hardware: [display driver used]

Please link to the driver on <https://crates.io> if applicable.

## Description of the problem/feature request/other

[description here] hello there .. I was kindly requesting info on the style function tried I got an error message that style is not found in embedded gaphics

## Test case (if applicable)

```rust
// Failing test case demonstrating the issue here
```unresolved import `embedded_graphics::style`
  --> src/main.rs:15:5
   |
15 |     style::{PrimitiveStyle, PrimitiveStyleBuilder},
   |     ^^^^^ could not find `style` in `embedded_graphics`

jamwaffles, 2024-01-17:
Hi there, `PrimitiveStyle` and `PrimitiveStyleBuilder` are available in `embedded_graphics::primitives`. That should get the imports working again.

Millan29, 2024-01-17:
thanks ... all good

closed 2024-01-17 by jamwaffles