embedded-graphics/embedded-graphics #758: A strange error "max I2C transfer is limited to 31 bytes"

trinhanhtuan, 2024-08-22:
- Version of embedded-graphics in use (if applicable): [0.8.0]
- Target: real hardware
- If using real hardware: sh1106 based OLED & ESP32 WROOM

Please link to the driver on [SH1106](https://github.com/jamwaffles/sh1106) if applicable.

## Description of the problem/feature request/other

Error while running a simple example as below :
```
!! A panic occured in '/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.18.0/src/i2c.rs', at line 1684, column 13:
PanicInfo {
    payload: Any { .. },
    message: Some(
        On ESP32 and ESP32-S2 the max I2C transfer is limited to 31 bytes,
    ),
    location: Location {
        file: "/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.18.0/src/i2c.rs",
        line: 1684,
        col: 13,
    },
    can_unwind: true,
    force_no_backtrace: false,
}

Backtrace:

0x400d1e75
0x400d1e75 - <esp_hal::i2c::I2C<T,esp_hal::Blocking> as embedded_hal::blocking::i2c::Write>::write
    at /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.18.0/src/i2c.rs:329
0x400d2dd8
0x400d2dd8 - sh1106::properties::DisplayProperties<DI>::draw
    at /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sh1106-0.5.0/src/properties.rs:96
0x400d193c
0x400d193c - sh1106::mode::graphics::GraphicsMode<DI>::flush
    at /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/sh1106-0.5.0/src/mode/graphics.rs:122
0x400d0e39
0x400d0e39 - sh1106_oled::__xtensa_lx_rt_main
    at /home/ESP32/sh1106-oled-02-esp32/src/main.rs:64
0x400d4a22
0x400d4a22 - Reset
    at /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/xtensa-lx-rt-0.16.0/src/lib.rs:70
0x400d40a7
0x400d40a7 - ESP32Reset
    at /root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.18.0/src/soc/esp32/mod.rs:93
0x3ffffffd
0x40079a81
0x400806b5
0x400806b5 - core::num::<impl u128>::trailing_zeros
    at /root/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/num/uint_macros.rs:150
0x40007c15
```

## Test case (if applicable)

```rust
#![no_std]
#![no_main]

use esp_backtrace as _;
use esp_hal::{
    clock::ClockControl,
    delay::Delay,
    gpio::Io,
    i2c::I2C,
    peripherals::Peripherals,
    prelude::*,
    system::SystemControl,
};

use embedded_graphics::{
    mono_font::{ascii::FONT_6X10, MonoTextStyleBuilder},
    pixelcolor::BinaryColor,
    prelude::*,
    text::{Baseline, Text},
};

use sh1106::{prelude::*, Builder, displaysize::DisplaySize};

#[entry]
fn main() -> !
{
    let peripherals = Peripherals::take();
    let system = SystemControl::new(peripherals.SYSTEM);
    let clocks = ClockControl::boot_defaults(system.clock_control).freeze();

    let delay = Delay::new(&clocks);
    let io = Io::new(peripherals.GPIO, peripherals.IO_MUX);

    // Create a new peripheral object with the described wiring
    // and standard I2C clock speed
    let i2c = I2C::new(
        peripherals.I2C0,
        io.pins.gpio21,
        io.pins.gpio22,
        100.kHz(),
        &clocks,
        None
    );

    let mut display: GraphicsMode<_> = Builder::new().connect_i2c(i2c).into();

    display.init().unwrap();
   display.flush().unwrap();
    display.clear();

    let text_style = MonoTextStyleBuilder::new()
                        .font(&FONT_6X10)
                        .text_color(BinaryColor::On)
                        .build();
    Text::with_baseline("Hello", Point::zero(), text_style, Baseline::Top).draw(&mut display).unwrap();

    match display.flush()
    {
        Ok(_) => esp_println::println!("Ok"),
        Err(_err) => panic!("Error: {:?}", _err)
    }

    loop{
        delay.delay_millis(1000);
    }
```

rfuest, 2024-08-22:
This is a bug in the driver crate and not e-g itself. You can try to decrease this value https://github.com/rust-embedded-community/sh1106/blob/798759149e6ec6efac1c21802215d7c441cfbbea/src/interface/i2c.rs#L46 to see if it fixes the issue. I would suggest to try a value of 16.

trinhanhtuan, 2024-08-23:
> This is a bug in the driver crate and not e-g itself. You can try to decrease this value https://github.com/rust-embedded-community/sh1106/blob/798759149e6ec6efac1c21802215d7c441cfbbea/src/interface/i2c.rs#L46 to see if it fixes the issue. I would suggest to try a value of 16.

Thank you for the suggestion. I patched with value of 16, no more error as earlier but it can't draw the text on the screen. My OLED is 128x64. As you can see in attachment, it seems only affect the first 16 (column) pixels. If I increase to 32 or above then the error comes back.
[image]

rfuest, 2024-08-23:
I think we should move this discussion to the `sh1106` repo. A proper fix will either need a constant generic or a feature flag to enable a shorter maximum transfer size. Setting the maximum size lower for all devices would unnecessarily degrade the performance for HAL's that support longer I2C transfers.

The reason that simply adjusting the `CHUNKLEN` doesn't work as I had expected is that the write address is set for each chunk (https://github.com/rust-embedded-community/sh1106/blob/798759149e6ec6efac1c21802215d7c441cfbbea/src/interface/i2c.rs#L66-L76) and simply adjusting the `CHUNKLEN` doesn't send the correct column addresses. To get the desired behaviour it would be necessary to first send the command that sets the address and then repeat this https://github.com/rust-embedded-community/sh1106/blob/798759149e6ec6efac1c21802215d7c441cfbbea/src/interface/i2c.rs#L78 command multiple times to write the 128 bytes for a page in smaller < 31 byte chunks.