embedded-graphics/embedded-graphics #759: Will embedded_graphics support asynchronous drawing?

skyarm, 2024-08-23:
- Version of embedded-graphics in use (if applicable): 0.8.1
- Target: RP2040
- If using real hardware: ST7789

## Description of the problem/feature request/other
Some display devices support DMA to transmit data, such as SPI, I2C, etc. In this way, data is transmitted asynchronously, which can greatly reduce the delay of data. It is hoped that embedded_graphics can support asynchronous drawing.

## Test case (if applicable)

```rust
// Failing test case demonstrating the issue here
```

rfuest, 2024-08-23:
At some point in the future e-g will probably get async support, but I'm not sure when this will happen. I've just started to work on e-g again and I need to work on fixing/finishing other things first. I don't think async will make it in the `0.9` release, but it would be nice to include async support in the one after that. But this also depends on other crates in the e-g ecosystem, like the display drivers and the `display-interface` crate, to support async.

Getting async support right will require some careful redesign of the `DrawTarget` API, because the current iterator based approach doesn't work well for DMA based transfers. I think the right approach is to first make changes to the API to allow for blocking DMA transfers and in a second step make the API optionally async.

Murmele, 2025-01-22:
@skyarm I did it in my project like this:

Create a buffer object for which you implement DrawTarget.
So clear, draw_iter and so on are not async. But you are writing just to a buffer.
This buffer object has a function "apply" which is async and sends out the buffer to the display.

jmjoy, 2025-09-15:
For MCUs with limited RAM, creating a framebuffer the size of the screen pixels isn’t very practical. I’m still more looking forward to the async version of e-g.