embedded-graphics/embedded-graphics #770: How to create larger fonts

Murmele, 2025-01-22:
I have seen the fonts in the repository, but perhabs they are all too small for my display. Is there any description how the raw files are generated to generate larger fonts?

rfuest, 2025-01-23:
One option is to use the not yet released `eg_font_converter` crate from the [`bdf` repo](https://github.com/embedded-graphics/bdf). This example shows how to use it in a build script to convert a BDF file to a `MonoFont`: https://github.com/embedded-graphics/bdf/blob/1527e39966d3d7fca698c29699752fbb66ae5bfa/eg-bdf-examples/build.rs#L26-L44

There are also some crates that provide additional fonts. Some of them are listed in the [external crates list](https://github.com/embedded-graphics/embedded-graphics?tab=readme-ov-file#additional-functions-provided-by-external-crates).

Murmele, 2025-01-23:
Thank you very much. I will have a look into them

closed 2025-01-23 by Murmele

Murmele, 2025-01-24:
My steps to get larger fonts:

1) Find a font in bdf format
2) Resize the font with bdfresize: I just created a cmake project for easier compilation: https://github.com/Murmele/bdfresize
3) use eg_font_converter  (See comment above)  to convert the rescaled font to a embedded graphics supported format