embedded-graphics/embedded-graphics #771: Feature request: Dashed stroke/fills

LaineZ, 2025-02-15:
Currently, the embedded graphics library seems to be not support dashed or dotted strokes and fills, similar to the `border-style: dashed` and `border-style: dotted` CSS properties. This feature would provide additional flexibility in styling elements, especially on `BinaryColor` displays.

[image]

inesvar, 2025-02-15:
Hello,

I'm interested in trying to implement this feature.
This would be my first contribution and I am currently discovering the code, I'll make an update here if I'm stuck.

rfuest, 2025-02-15:
This would be a great addition. Implementing this for all primitives could get complicated, e.g. for lines with > 1px stroke width and line joins, but we could always start with a subset of supported primitives and expand support to additional types later.

@inesvar: Thanks for trying to implement this, let us know if we can help in any way.

inesvar, 2025-02-15:
Hello again, I need some advice on how to test my future commits.

I get a compilation error when trying to run an example from the embedded-graphics/examples repo using the embedded-graphics source code instead of the crates.io package.

`eg-0.8/Cargo.toml` was changed from
```
embedded-graphics = "0.8.1"
```
to
```
# embedded-graphics = "0.8.1"
# embedded-graphics = { path = "../../embedded-graphics", version = "0.8.1"} # this doesn't work either
# embedded-graphics = { git = "https://github.com/embedded-graphics/embedded-graphics.git", tag = "embedded-graphics-v0.8.0" } # same
embedded-graphics = { git = "https://github.com/embedded-graphics/embedded-graphics.git", tag = "embedded-graphics-v0.8.1" }
```

However while the example "primitives-stroke" was compiling in the first case, I now get a compilation error:

[image]

```
cargo run --example primitives-stroke
....
error[E0599]: the function or associated item `new` exists for struct `SimulatorDisplay<Rgb888>`, but its trait bounds were not satisfied
   --> examples/primitives-stroke.rs:58:51
    |
58  |     let mut display = SimulatorDisplay::<Rgb888>::new(Size::new(512, 512));
    |                                                   ^^^ function or associated item cannot be called on `SimulatorDisplay<Rgb888>` due to unsatisfied trait bounds
    |
   ::: /home/inesv/.cargo/git/checkouts/embedded-graphics-1bd1f33e364edb1f/f0dce19/core/src/pixelcolor/rgb_color.rs:237:1
    |
237 | rgb_color!(Rgb888, RawU24, u32, Rgb = (8, 8, 8));
    | ------------------------------------------------ doesn't satisfy `_: From<BinaryColor>` or `_: PixelColor`
    |
    = note: the following trait bounds were not satisfied:
            `Rgb888: embedded_graphics_core::pixelcolor::PixelColor`
            `Rgb888: From<embedded_graphics_core::pixelcolor::binary_color::BinaryColor>`
```

Rustdoc says these traits are implemented, and their source code is present.

I use the following toolchain:
```
stable-x86_64-unknown-linux-gnu (default)
rustc 1.84.1 (e71f9a9a9 2025-01-27)
```

rfuest, 2025-02-15:
You need to make sure that all dependencies are changed to the GIT version. In your case the simulator will still use the crates.io version of e-g 0.8. Take a look at cargo's patch feature: https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#the-patch-section. This is also used in the `eg-next` version of the examples repo: https://github.com/embedded-graphics/examples/blob/baced2aeeebc5444a2ec6ad064a38a51589f8567/eg-next/Cargo.toml#L17-L22

inesvar, 2025-02-16:
Thank you for your answer.

The `eg-next` version of the examples repo doesn't compile either (more precisely, tinytga and tinybmp don't compile with the most recent version of embedded-graphics). However I finally managed to change `eg-next/Cargo.toml` so that I can run the examples with my fork of embedded-graphics (by removing tinybmp and tinytga dependencies and pointing all embedded-graphics(-core) dependencies to my fork).

I will start coding the actual feature next week-end. I'm thinking of starting with the dotted line - since filled circles are already implemented - for the rectangle. For the line joints, I noticed that in browsers (firefox) there are dots in each corner, and then the number of dots on each line is computed by taking the border width into account.

inesvar, 2026-09-30:
Hey, it has been a while !
(I should have left a message when I checked out, sorry)

About a year ago I suddenly noticed that I didn't update the `StyledPixels` trait with the new dotted style. Looking into it, I realized updating the `StyledPixels` implementations to support the dotted style would be a pain and I got demotivated (Am I just adding an ugly code that won't get used, uses floats and doesn't integrate well with the existing code ? 😢).

Anyway I have fixed my other personal project since, and I have time and motivation again :)
I think it would be a waste to not reap the full benefits of the "DottedLineIterator". #828 is open to deduplicate dotted rectangle borders and dotted lines, then I'll look into dotted Polyline (then dotted circles, it's half-implemented from last year :D).

@rfuest I saw your comment in issue #827

> The current PrimitiveStyle is getting to big already and I'm planning to move the advanced features, like dotted strokes, into a separate style.

Nice idea !
Could it mean that `StyledPixels` would be implemented separately for the normal styles and the dotted style ? 👀